floripasat/eps #17: Improve how to define the battery charge capability

mgm8, 2019-06-29:
**Is your feature request related to a problem? Please describe.**
Now, when the batteries are changed to a different model, with a different charge capacity, it's hard to define the new maximum battery charge.

**Describe the solution you'd like**
An easy way to define the battery capacity, maybe a ```#define``` directive in a header file (maybe a ```config.h``` file) defining the maximum charge in Ah.

**Describe alternatives you've considered**
None.

**Additional context**
This issue is related to #16.

mgm8, 2019-08-02:
68534e2e5b88b88891e824ac72928418bf440684

mgm8, 2019-08-02:
abc2fe86fe7fa0c70ceede85aef48047ec82eaae

ramonborba, 2019-08-13:
Issue resolved

closed 2019-08-15 by mgm8